chris-jh/mt32-pi-midi-hat #1: OSHPark shows board bottom as white

leolobato, 2021-02-05:
I'm not experienced at all with producing/ordering PCBs but am looking forward to assembling this project.

I've tried uploading the Gerbers_v1_7.zip file to OSHPark and they showed an issue at the board bottom (see attached).

Is this something that is easily fixable?

[image]

chris-jh, 2021-02-05:
the gerbers are made to JLCPCB Specification, so maybe this is the issue ?

I think you will need to use KiCad to make new Gerbers to OSHParks Specifications, or order from JLCPCB

leolobato, 2021-02-05:
Thank you! I'll check JLCPCB.

closed 2021-02-05 by leolobato