chris-jh/mt32-pi-midi-hat #2: 1.7 on oshpark bottom white?

b0rnwr0ng, 2021-02-16:
Is the template that is uploaded correct?  Seems to render incorrectly

b0rnwr0ng, 2021-02-18:
Reading through everything I see it was made with recommends with jlxpcb.  I was having issues there so went another route.  Will let you know how they turn out, ordered from another site as well.

Also got 1.5 working no issues

chris-jh, 2021-02-18:
I think it may have been the Midi-Din connector being right on the edge so i have moved it and produced another gerber.zip.

b0rnwr0ng, 2021-02-18:
That worked!  I'll post how the 1.7 came out if I'm able to.

closed 2021-02-18 by b0rnwr0ng

b0rnwr0ng, 2021-03-01:
I didn't realize there was a closed issue already on this.  I got back the 1.7 boards and seem ok, although hace a ton 1.7.1s on the way back