chris-jh/mt32-pi-midi-hat #3: Tracer is very close to mounting screw.

skadarnold, 2021-03-08:
Minor issue where the tracer next to this GPIO pin can be exposed when tightening mounting screws to raspberry pi case. Recommended using a washer or nylon screws when using version 1.7 of this hat. Causes the raspberry pi to short when turning on. Did not cause any damage from what I can tell as removing the screw allowed everything to work as it should. It is also worth mentioning that I tightened as normal, and did not necessarily over tighten this screw. 

[image]

chris-jh, 2021-06-13:
I will move it, when i get around, good call. Although i used nylon screws, so never noticed.

chris-jh, 2021-06-13:
fixed

closed 2021-06-13 by chris-jh