chris-jh/mt32-pi-midi-hat #5: J6 Pinout

appiah4, 2021-07-14:
Hi building one of these for use as a hat.  I will use on board microswitches for now but hope to design a case with external buttons eventually.  J6 seems to be for this purpose, but I can't seem to find the pinout and how to connect external buttons to the device in the documentation.  Can you assist? Thanks.

doccaz, 2022-04-15:
I just assembled mine and was also wondering the same thing. Is J6 a breakout for the switches, or is it for connecting to the user port of a MiSTer FPGA?

chris-jh, 2022-08-03:
Sorry been a while, been really busy with work.

Yeah the J6 is just a break out of the buttons, incase you want to put in case like i did.

My board was designed before the MiSTer connection, i thought about adding it. but i don't think there is a need as there are other boards that are better suited to MiSTer.

DOSSound, 2024-09-02:
Hello... sorry to bother you... could you tell me the Pinout of the J6 connector? I intend to use an Encoder and 2 buttons, but I don't know which pins to connect it to.
And also, could you tell me if you use a Case for the MT32 pi and share the STL with me... Please
Thank you!!!

doccaz, 2024-09-07:
> Hello... sorry to bother you... could you tell me the Pinout of the J6 connector? I intend to use an Encoder and 2 buttons, but I don't know which pins to connect it to. And also, could you tell me if you use a Case for the MT32 pi and share the STL with me... Please Thank you!!!

Actually I just pinned it out with a multimeter, so it's:

1 - MT32/Soundfont
2 - Synth
3 - Vol Down
4 - Vol Up
5 - not connected (I guess it was for the encoder button that wasn´t  implemented?)
6 - Ground

I haven't used mine with a case, I ended up building some MT32Pi hats for the MiSTer since that was what I was looking for at the time I asked this. It's not my project, but I guess one could be made that holds both the Pi and the Hat. If I do one, I'll post it at https://www.thingiverse.com/docca/designs.

DOSSound, 2024-09-09:
Thank you!!!
How could I put a Rotary Encoder on it?

doccaz, 2024-09-09:
> Thank you!!! How could I put a Rotary Encoder on it?

Just follow this wiki: https://github.com/dwhinham/mt32-pi/wiki/Control-surface

You need to define a simple_encoder scheme, and connect the encoder to the mentioned pins (increase/decrease volume as encoder CLK or DAT).