chris-jh/mt32-pi-midi-hat #8: Why are there two sets of buttons?

sofakng, 2021-12-31:
Why are there two sets of buttons?

Am I correct that if you install the 128x32 LCD (long horizontal) then you should not install the 4 square button layout in the corner of the board?  (the LCD almost fits but hits the corner of the Synth button)

Similarly, if you install the 128x64 LCD that you should not install the horizontal row of buttons?  (it looks like the LCD would sit on top of the buttons if I understand the layout correctly which seems obviously wrong)

doccaz, 2022-04-15:
I just assembled mine with a 128x32 OLED and only after assembling everything I noticed that the buttons were duplicated :)

The 128x32 OLED is so thin that it doesn't hit any of the buttons. But adding the 128x64 OLED surely will hit the horizontal row.
Maybe would be useful to add some lettering to the board along the lines of "if you use the 128x64 OLED, don't install these switches".

chris-jh, 2022-08-03:
Correct, incase you use the different oled screen sizes which have different locations for connecting to the board.

closed 2022-08-03 by chris-jh