chris-jh/mt32-pi-midi-hat #15: Reset Sound

DOSSound, 2024-12-02:
Hello.. I have completed the assembly of this device...
When testing it, I noticed that every so often (randomly) the audio cuts off (something similar to an audio reset).
I'm using it with a keyboard connected by MIDI cable and it sounds SUPER... but every once in a while it's like I stop pressing the keys the way the audio cuts out...
Do you know if there is any reason for this? 
I have tried changing values ​​for the number of notes, sampling frequency and things like that... I have not touched the Baud Rate and I have WIFI+usb+ftp enabled.
The sound sources I'm using range from less than 1mb to a couple hundred (I don't think this is the cause of the problem)

Greetings!

chris-jh, 2024-12-05:
Hi, i am unsure.

You might need to refer to the mt32Pi project as it was based on the schematic provided by Dale (https://github.com/dwhinham/mt32-pi/wiki/GPIO-MIDI-interface)

Have you search the issues section on the main mt32-pi project page for anything similar ?

I am unfamiliar with the software side, my pi hat is still running a very old version 😂

Unless faulty DAC ?

DOSSound, 2024-12-11:
Thank you! In the first instance, I assume that it is a problem with my keyboard since I have tested your implementation with the Bulky Midi Play Project (which is a .mid file player that needs an external device that has the synthesizer) and the problem mentioned above did not occur. . 
I haven't been able to get another keyboard or any other way to check this...
When I have the confirmation I'll tell you...
Greetings!